DCS-Skunkworks/dcs-bios-arduino-library #2: Analog Multipos

tekcreations, 2020-11-23:
Heya, I was working with Warlord to add my analogMultipos code  into the library and hadn't finished testing yet, you forked it into this It doesn't seem to work. If you can  reach out to me tekadept@tekcreations.space can try and sort it out.

talbotmcinnis, 2020-11-24:
Let's figure it out!  Yes I got the work he had done and merged it in, but did not test it on a device so it's very possible there is still work to do. 

I'll send you an email.

talbotmcinnis, 2020-11-27:
Established via email that the control is not emitting any commands.  Requested original source code to compare, and will setup a test bench myself.

talbotmcinnis, 2020-11-28:
Tweaked a bit, and then tested on my hardware and this branch works with the following in my sketch.  Give it a try and let me know:
DcsBios::AnalogMultiPos testAnalogMultiPos("TEST_CMD", 44, 10);

https://github.com/talbotmcinnis/dcs-bios-arduino-library/tree/Fix-AnalogMultipos

I can't imagine this working well without a hysteresis.  I expect it to frequently flip between states unless you have some detents on the physical.  If that's the case I can likely come up with a solution for that.

talbotmcinnis, 2020-11-28:
Comment made on the issue in github.  Is it possible to communicate through
that so we have a record?

tekcreations, 2020-12-05:
I can happily report it is working for me as well with my existing code :)

talbotmcinnis, 2020-12-05:
Excellent stuff!  Release 0.3.1 created.

closed 2020-12-05 by talbotmcinnis