DCS-Skunkworks/dcs-bios-arduino-library #7: Possible STM32 support - pull available on original arduino library repo

virtual812, 2021-01-21:
Hi there,
Hope I'm putting the right info in the right place.
I get lost in github sometimes.

This is the pull on the oridinal DCS BIOS arduino lib repo...
https://github.com/dcs-bios/dcs-bios-arduino-library/pull/27

The Arduino mega 2560 has been a bit of a show stopper due to running out of ram.
There are a few STM 32 dev boards that would that the 8KM SRAM of the arduino to ~ 96KB.

I might only be causing a hassle, but i hope not.
What I'm doing with DCS BIOS seems to be different to (what i think) it's intended purpose is.
I have an LCD and ma driving different content to it depending on the pit i'm in.
This is working great, but have run out of room (moreso RAM) on the Mega 2560.
One example of issue is the PVI-800 output for the Ka-50 - the commas are not part of the string, but when writing to the LCD i need them in, and with some creative code arduino side a mate helped this happen, but litle issues like this are piling up and we've hit the SRAM limit for the code,

Happy for anyone to take and act on this possible STM32 support as seen fit.
If anyone is available i'd also probably love to ask more questions, but I'm sure you're busy so i'll leave it at this for now.

Last thing, do you accept donations, can u buy you a coffee or a beer?
Happy to send a bit of appreciation for an active project.

talbotmcinnis, 2021-01-21:
Hey.  Thanks for the intro/backstory. I confess I know very little about the STM32, but there isn't much actual code in the dcs-bios library itself, and having an existing PR helps a lot.  

I'm happy to start tinkering and see how it looks, then I'll have a better idea of the scope.  Do you have an stm32?  You might need to be the official field-tester :)

It's nice to know the efforts are appreciated, that's enough for me :). I'll take a look this weekend and get back to you.

virtual812, 2021-01-21:
I do have an STM32, a 'Blue Pill' which is one of the cheaper STM32 dev
boards.
Usually under ~USD$3 on AliExpress, 32bit,  72MHz Arm Cortex M4, onboard
12bid ADC's etc..etc..

[image: image.png]
Capture from https://stm32-base.org/boards/STM32F103C8T6-Blue-Pill
Click boards up the top to see the variety of others.
If you're interested yourself there are some listings that have a 5 pack of
blue pills and include an STLINK programmer for pretty cheap.
I'd be happy to order a package and send them your way if you think they
might be of any use either for DCS BIOS, HID joy or whatever else you could
imagine.

I'm more familiar with AVR from using them as Arduinos.

Came across a project called FreeJoy, firmware and config software for
using an STM32 'Blue Pill' as a HID joystick, absolutely fantastic work
there.
https://github.com/FreeJoy-Team/FreeJoy

I'm interested in the variants that have ~ 192KB SRAM, they are around $12
on Aliexpress if i recall.

Happy to test, I am leaving on a short holiday soon and doing a lot of work
tweaking my electronics bench.
Once I'm back from that I should be well setup and firing at full power.
Cheers,
Virt

talbotmcinnis, 2021-01-26:
So I rolled up my sleeves to get started on this today, read the PR and it looked simple enough.  To my surprise when I went to get started, all of those changes are already included in this branch.  Just define DCSBIOS_FOR_STM32 before including the libraries.  I don't recall what branch I started my fork from, but I did download a BSP for Arduino IDE and compiled without error.
I can't do a full test without the hardware, but can you try just pulling this master and seeing if it works for you?

virtual812, 2021-02-11:
Wow, sorry for leaving this so long.
Had a short holiday, now a house full of sick people and, but i have made massive strides on making room on my bench.
I have about 3 other projects i need to knock over then will pull out the DCS BIOS hardware and cover the desk in it.
Love to make a mess!!
I don't have git installed locally, never used it, just done everything web based so far.
I should be able to just do all this in the Arduino IDE as i have previously, just with the library from this repo instead right??
Sorry for the silly questions, programming isn't my thing usually, I just do what i can where i can and learn a little along the way.

talbotmcinnis, 2021-02-11:
Hey. Yes O totally get these projects going in bursts then idling.  All good!

You don’t need to get fully into git to use the repo. You can just download the code as a zip and then yes it is like any other Arduino library.  I have packaged releases that would be the best place to start.  If you get any build errors, send me the full text and we will go forwards from there.

talbotmcinnis, 2021-05-31:
Did you have a chance to test this?  If so we can close this issue, and of course open new ones if things pop up.

virtual812, 2021-06-02:
I do apologise, had to pack up my lab for a bit.
I have just set everything up again and will give it a crack.

virtual812, 2021-06-02:
So apologies for the delay, am all geared up, have achieved the following, but also hit a snag...
STM Blue pill on the bench.
Latest Arduino IDE installed.
STM32duino boards manager lib installed.
Blink running on the board uploaded via the STLink.
DCSFlightpanels / dcs-bios  installed
DCS-BIOS Control Reference Page Chrome plugin installed
DCS BIOS tested and working via the Control ref page
Ka-50 PVI-800 display co-ords show in the plugin
Can switch on/off battery and adjust Abriss brightness in DCS with result in the Control Ref Plugin
Can Adjust the same as above in the Control Ref Plugin and have the result appear in DCS.

The last hurdle... the Arduino library.
In the Arduino IDE i go Sketch > Include Library > Add .ZIP library and select the dcs-bios-arduino-library-0.3.3.zip file i get the following error...
`java.io.FileNotFoundException: C:\Users\v81\AppData\Local\Temp\arduino_481168\dcs-bios-arduino-library-0.3.3(2)856560\examples\Button_Matrix_Example\Button_Matrix_Example.ino (The system cannot find the path specified)
	----- plus a bunch of other stuff, can elaborate if it helps
`
Is the library a valid arduino IDE lib that can be imported this way?
Anyway, I installed the library manually and went to fire up the Master Caution example.
Added #define DCSBIOS_FOR_STM32 and changed LED to PC13 and got the following error when trying to compile...
Wall incoming

`C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\v81\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\v81\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\v81\Documents\Arduino\libraries -fqbn=stm32duino:STM32F1:genericSTM32F103C:device_variant=STM32F103C8,upload_method=DFUUploadMethod,cpu_speed=speed_72mhz,opt=osstd -ide-version=10815 -build-path C:\Users\v81\AppData\Local\Temp\arduino_build_994620 -warnings=none -build-cache C:\Users\v81\AppData\Local\Temp\arduino_cache_635050 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.stm32tools.path=C:\Users\v81\AppData\Local\Arduino15\packages\stm32duino\tools\stm32tools\2021.5.31 -prefs=runtime.tools.stm32tools-2021.5.31.path=C:\Users\v81\AppData\Local\Arduino15\packages\stm32duino\tools\stm32tools\2021.5.31 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\v81\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.arm-none-eabi-gcc-4.8.3-2014q1.path=C:\Users\v81\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -verbose C:\Users\v81\AppData\Local\Temp\arduino_modified_sketch_311120\MasterCaution.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\v81\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\v81\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\v81\Documents\Arduino\libraries -fqbn=stm32duino:STM32F1:genericSTM32F103C:device_variant=STM32F103C8,upload_method=DFUUploadMethod,cpu_speed=speed_72mhz,opt=osstd -ide-version=10815 -build-path C:\Users\v81\AppData\Local\Temp\arduino_build_994620 -warnings=none -build-cache C:\Users\v81\AppData\Local\Temp\arduino_cache_635050 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.stm32tools.path=C:\Users\v81\AppData\Local\Arduino15\packages\stm32duino\tools\stm32tools\2021.5.31 -prefs=runtime.tools.stm32tools-2021.5.31.path=C:\Users\v81\AppData\Local\Arduino15\packages\stm32duino\tools\stm32tools\2021.5.31 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\v81\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.arm-none-eabi-gcc-4.8.3-2014q1.path=C:\Users\v81\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -verbose C:\Users\v81\AppData\Local\Temp\arduino_modified_sketch_311120\MasterCaution.ino
Using board 'genericSTM32F103C' from platform in folder: C:\Users\v81\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2021.5.31
Using core 'maple' from platform in folder: C:\Users\v81\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2021.5.31
Detecting libraries used...
"C:\\Users\\v81\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -DDEBUG_LEVEL=DEBUG_NONE -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -fno-use-cxa-atexit -DBOARD_generic_stm32f103c -DVECT_TAB_ADDR=0x8002000 -DERROR_LED_PORT=GPIOC -DERROR_LED_PIN=13 -w -x c++ -E -CC -mcpu=cortex-m3 -DF_CPU=72000000L -DARDUINO=10815 -DARDUINO_GENERIC_STM32F103C -DARDUINO_ARCH_STM32F1 -DSERIAL_USB -DGENERIC_BOOTLOADER -DMCU_STM32F103C8 -mthumb -march=armv7-m -D__STM32F1__ -DMCU_STM32F103C8 -mthumb -march=armv7-m -D__STM32F1__ "-IC:\\Users\\v81\\AppData\\Local\\Arduino15\\packages\\stm32duino\\hardware\\STM32F1\\2021.5.31\\system/libmaple" "-IC:\\Users\\v81\\AppData\\Local\\Arduino15\\packages\\stm32duino\\hardware\\STM32F1\\2021.5.31\\system/libmaple/include" "-IC:\\Users\\v81\\AppData\\Local\\Arduino15\\packages\\stm32duino\\hardware\\S [... cut ...]
exit status 1
Error compiling for board Generic STM32F103C series.
`

Would love to get it going as we've run out of SRAM on the ArduinoMEGA2560 and the STM looks like it might give us more than double (20KiB vs 8KiB)
Not much of a code person myself so don't understand 99% of it.
Any thoughts or advice appreciated, happy to provide whater i can to be helpful.
I have in the last few days got the lab 90% setup again, finishing touches to come, but can at least put thoughts into action in it's current state.

talbotmcinnis, 2021-06-02:
> C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\v81\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\v81\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\v81\Documents\Arduino\libraries -fqbn=stm32duino:STM32F1:genericSTM32F103C:device_variant=STM32F103C8,upload_method=DFUUploadMethod,cpu_speed=speed_72mhz,opt=osstd -ide-version=10815 -build-path C:\Users\v81\AppData\Local\Temp\arduino_build_994620 -warnings=none -build-cache C:\Users\v81\AppData\Local\Temp\arduino_cache_635050 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.stm32tools.path=C:\Users\v81\AppData\Local\Arduino15\packages\stm32duino\tools\stm32tools\2021.5.31 -prefs=runtime.tools.stm32tools-2021.5.31.path=C:\Users\v81\AppData\Local\Arduino15\packages\stm32duino\tools\stm32tools\2021.5.31 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\v81\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.arm-none-eabi-gcc-4.8.3-2014q1.path=C:\Users\v81\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -verbose C:\Users\v81\AppData\Local\Temp\arduino_modified_sketch_311120\MasterCaution.ino C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\v81\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\av [... cut ...] : error: 'UDR0' was not declared in this scope UDR0 = *c++; // write byte to TX buffer ^ Using library dcs-bios-arduino-library-0.3.3 at version 0.3.3 in folder: C:\Users\v81\Documents\Arduino\libraries\dcs-bios-arduino-library-0.3.3 exit status 1 Error compiling for board Generic STM32F103C series.

I found out I have been packaging that lib wrong when I zip it, so the first error will be fixed.  Good job on the workaround.

I will try to recreate those compilation errors and get back to you.

Alan-JH, 2021-06-05:
Hello,

I just found this thread. I am in a similar position, trying to use an STM32 bluepill to run a UFC.
I was able to get the code compiled and running with this workaround: https://forums.eagle.ru/topic/171932-dcs-bios-support-for-stm32-board/
Though for some reason the hub gets stuck on "connecting..." when I try to connect to the COM port in the web interface.

virtual812, 2021-06-06:
Alan, are you using the original DCS-BIOS or the Flight Panels version?
I'm looking to move back to the old one as the web interface for the conrtol references is easier to use and not reliant on having Google Chrome installed.

Alan-JH, 2021-06-06:
Sorry I didn't realize there was another version. I'm using the original DCS-BIOS.

talbotmcinnis, 2021-06-06:
The versions work interchangeably so if you want to jump onto this fork I could help with future issues.  I have one user testing stm32 support.

Alan-JH, 2021-06-08:
Hey, 

I found some more time to get my code working today. I hadn't realized I was on an older fork of the arduino library when I commented earlier. Sorry about that. 

I noticed on the old fork, with both the original DCS-BIOS and the FlightPanels version, the COM port was silent after the cockpit loaded in, so I switched over to this newer version, and that solved it. I got rid of the old workaround, and defined DCSBIOS_FOR_STM32 and DCSBIOS_DEFAULT_SERIAL, and it compiled and uploaded, and interfaced with DCS World perfectly. I still have a couple issues to work out, but its almost certainly my own code.

Anyways, if you'd like me to test anything further, I'd be happy to. I'm running an STM32F103C8T6 with a Maple Mini bootloader, so that I can upload to it via usb.

virtual812, 2021-06-11:
Talbot, Is there any particular benefit to using the Flight Panels version of DCS BIOS?
I find the chrome plugin it relies on for the control refs annoying, really wanting to steer clear of Chrome.

Alan, how did you get around the issues that I had with compiling for STM32?
Are you using the "STM32duino" board package in the arduino IDE or something else?
I'm using an ST link, but could take a look at this Maple bootloader you mention.
I'd love to just get a successful 'hello world' style compile (i think the master caution light is considered a fairly typical hello world for DCS-BIOS).

Once you had it compiled how does DCS BIOS talk to the board, is it somehow configuring the STM32's USB port to be a USB/serial device or are you using one of the UARTS on the board?

For my project i'm using a serial driven HMI display (a 'Nextion' display) so i need a serial port on the micro not tied up by DCS BIOS.
So far I've been using an Arduino MEGA 2560 for it's good amount of SRAM and serial ports, but i don't need all the I/O on it.

talbotmcinnis, 2021-06-11:
2 sides to that answer:

The arduino side and the pc side.  Both are interchangeable so you can use either with the hub or fp fork and vice versa.  I am partial to the ongoing work one this fork.  

The pc side major advantage is that it is still supported.  Numerous fixes to new aircraft, and the more obscure functions of old aircraft.  Now if the hub version does provide all the functions you need to map, you are lucky and it does make sense to enjoy the better ux.  But if you find an issue you are sol short of self-fixing perhaps.

Ultimately 100% your call and whatever works for you is the right answer.

Alan-JH, 2021-06-11:
I didn't really have any issues compiling. I did forget to define DCSBIOS_DEFAULT_SERIAL at one point, which caused compile errors, but they don't match what you have. I am using STM32duino, yes. I had basically followed [this](https://www.youtube.com/watch?v=saolJ_7E7hc) guide, as I am very new to STM32 myself. As for USB behavior, the port switches between modes after the file is uploaded, as I understand it, so the usb port can be used to upload, and interface.

talbotmcinnis, 2021-07-24:
Sounds like because this CAN work, so I'm going to close this issue.  Of course we can open new ones if we have specific issues to investigate.

closed 2021-07-24 by talbotmcinnis

speedbird5, 2021-11-10:
Hi, I cannot get a simple DCS-BIOS sketch to compile on my STM32F103C8 blue pill.
I can however compile and upload the simple Blink sketch going and read potentiometer outputs etc.
I used the STM32duino board definitions.
I have installed your Arduino library version 3.6 .
Hers is the sketch code:

`/* use '#define DCSBIOS_DEFAULT_SERIAL' instead if your Arduino board
 *  does not feature an ATMega328 or ATMega2650 controller.
 */
#define DCSBIOS_DEFAULT_SERIAL
#define DCSBIOS_FOR_STM32
#include "DcsBios.h"

/* Declare a Master Caution Reset button on pin 10 */
DcsBios::Switch2Pos ufcMasterCaution("UFC_MASTER_CAUTION", PB4);
/* Make the LED connected to pin 13 into a Master Caution Light */
DcsBios::LED masterCaution(0x1012, 0x0800, PC13);

void setup() {
  DcsBios::setup();
}

void loop() {
  DcsBios::loop();
}`

I get the following errors on compile:
In file included from C:\Users\Peter\Documents\Arduino\libraries\dcs-bios-arduino-library-0.3.6\src/DcsBios.h:130,
                 from C:\Users\Peter\AppData\Local\Temp\arduino_modified_sketch_800786\MasterCaution.ino:6:
C:\Users\Peter\Documents\Arduino\libraries\dcs-bios-arduino-library-0.3.6\src/internal/RotarySwitch.h: In member function 'void DcsBios::RotarySwitchT<pollIntervalMs, stepsPerDetent>::pollInput()':
C:\Users\Peter\Documents\Arduino\libraries\dcs-bios-arduino-library-0.3.6\src/internal/RotarySwitch.h:48:54: error: no matching function for call to 'min(int, signed char&)'
   48 |     switchValue_ = min(switchValue_+1, maxSwichValue_);
      |                                                      ^
In file included from c:\users\peter\appdata\local\arduino15\packages\stmicroelectronics\tools\xpack-arm-none-eabi-gcc\10.2.1-1.1\arm-none-eabi\include\c++\10.2.1\algorithm:62,
                 from C:\Users\Peter\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.1.0\cores\arduino/wiring_constants.h:26,
                 from C:\Users\Peter\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.1.0\cores\arduino/wiring.h:34,
                 from C:\Users\Peter\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.1.0\cores\arduino/Arduino.h:36,
                 from sketch\MasterCaution.ino.cpp:1:
c:\users\peter\appdata\local\arduino15\packages\stmicroelectronics\tools\xpack-arm-none-eabi-gcc\10.2.1-1.1\arm-none-eabi\include\c++\10.2.1\bits\stl_algo.h:3474:5: note: candidate: 'template<class _Tp, class _Compare> constexpr _Tp std::min(std::initializer_list<_Tp>, _Compare)'
 3474 |     min(initializer_list<_Tp> __l, _Compare __comp)
      |     ^~~
c:\users\peter\appdata\local\arduino15\packages\stmicroelectronics\tools\xpack-arm-none-eabi-gcc\10.2.1-1.1\arm-none-eabi\include\c++\10.2.1\bits\stl_algo.h:3474:5: note:   template argument deduction/substitution failed:
In file included from C:\Users\Peter\Documents\Arduino\libraries\dcs-bios-arduino-library-0.3.6\src/DcsBios.h:130,
                 from C:\Users\Peter\AppData\Local\Temp\arduino_modified_sketch_800786\MasterCaution.ino:6:
C:\Users\Peter\Documents\Arduino\libraries\dcs-bios-arduino-library-0.3.6\src/internal/RotarySwitch.h:48:54: note:   mismatched types 'std::initializer_list<_Tp>' and 'int'
   48 |     switchValue_ = min(switchValue_+1, maxSwichValue_);
      |                                                      ^
In file included from c:\users\peter\appdata\local\arduino15\packages\stmicroelectronics\tools\xpack-arm-none-eabi-gcc\10.2.1-1.1\arm-none-eabi\include\c++\10.2.1\algorithm:62,
                 from C:\Users\Peter\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.1.0\cores\arduino/wiring_constants.h:26,
                 from C:\Users\Peter\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.1.0\cores\arduino/wiring.h:34,
                 from C:\Users\Peter\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.1.0\cores\arduino/Arduino.h:36,
                 from sketch\MasterCaution.ino.cpp:1:
c:\users\peter\appdata\local\arduino15\packages\stmicroelectronics\tools\xpack-arm-none-eabi-gcc\10.2.1-1.1\arm-none-eabi\include\c++\10.2.1\bits\stl_algo.h:3468:5: note: candidate: 'template<class _Tp> constexpr _Tp std::min(std::initializer_list<_Tp>)'
 3468 |     min(initializer_list<_Tp> __l)
      |     ^~~
c:\users\peter\appdata\local\arduino15\packages\stmicroelectronics\tools\xpack-arm-none-eabi-gcc\10.2.1-1.1\arm-none-eabi\include\c++\10.2.1\bits\stl_algo.h:3468:5: note:   template argument deduction/substitution failed:
In file included from C:\Users\Peter\Documents\Arduino\libraries\dcs-bios-arduino-library-0.3.6\src/DcsBios.h:130,
                 from C:\Users\Peter\AppData\Local\Temp\arduino_modified_sketch_800786\MasterCaution.ino:6:
C:\Users\Peter\Documents\Arduino\libraries\dcs-bios-arduino-library-0.3.6\src/internal/RotarySwitch.h:48:54: note:   mismatched types 'std::initializer_list<_Tp>' and 'int'
   48 |     switchValue_ = min(switchValue_+1, maxSwichValue_);
      |                                                      ^
In file included from c:\users\peter\appdata\local\arduino15\packages\stmicroelectronics\tools\xpack-arm-none-eabi-gcc\10.2.1-1.1\arm-none-eabi\include\c++\10.2.1\algorithm:61,
                 from C:\Users\Peter\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.1.0\cores\arduino/wiring_constants.h:26,
[... cut ...]
C:\Users\Peter\Documents\Arduino\libraries\dcs-bios-arduino-library-0.3.6\src/internal/RotarySwitch.h:48:54: note:   deduced conflicting types for parameter 'const _Tp' ('int' and 'signed char')
   48 |     switchValue_ = min(switchValue_+1, maxSwichValue_);
      |                                                      ^
Multiple libraries were found for "Servo.h"
 Used: C:\Users\Peter\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.1.0\libraries\Servo
 Not used: C:\Program Files (x86)\Arduino\libraries\Servo
exit status 1
Error compiling for board Generic STM32F1 series.