DCS-Skunkworks/dcs-bios-arduino-library #19: STM32 Compile Issues

speedbird5, 2021-11-10:
Hi, I cannot get a simple DCS-BIOS sketch to compile on my STM32F103C8 blue pill.
I can however compile and upload the simple Blink sketch and another sketch which reads potentiometer outputs etc with no issues.
I used the STM32duino board definitions.
I have installed your Arduino library version 3.6  from here: https://github.com/talbotmcinnis/dcs-bios-arduino-library/releases/download/0.3.6/dcs-bios-arduino-library-0.3.6.zip
I am using the Flight Panels DCS BIOS code.

Hers is the sketch code:

#define DCSBIOS_DEFAULT_SERIAL
#define DCSBIOS_FOR_STM32
#include "DcsBios.h"

/* Declare a Master Caution Reset button on pin 10 */
DcsBios::Switch2Pos ufcMasterCaution("UFC_MASTER_CAUTION", PB4);
/* Make the LED connected to pin 13 into a Master Caution Light */
DcsBios::LED masterCaution(0x1012, 0x0800, PC13);

void setup() {
  DcsBios::setup();
}

void loop() {
  DcsBios::loop();
}

I get the following errors on compile:

In file included from C:\Users\Peter\Documents\Arduino\libraries\dcs-bios-arduino-library-0.3.6\src/DcsBios.h:130,
                 from C:\Users\Peter\AppData\Local\Temp\arduino_modified_sketch_800786\MasterCaution.ino:6:
C:\Users\Peter\Documents\Arduino\libraries\dcs-bios-arduino-library-0.3.6\src/internal/RotarySwitch.h: In member function 'void DcsBios::RotarySwitchT<pollIntervalMs, stepsPerDetent>::pollInput()':
C:\Users\Peter\Documents\Arduino\libraries\dcs-bios-arduino-library-0.3.6\src/internal/RotarySwitch.h:48:54: error: no matching function for call to 'min(int, signed char&)'
   48 |     switchValue_ = min(switchValue_+1, maxSwichValue_);
      |                                                      ^
In file included from c:\users\peter\appdata\local\arduino15\packages\stmicroelectronics\tools\xpack-arm-none-eabi-gcc\10.2.1-1.1\arm-none-eabi\include\c++\10.2.1\algorithm:62,
                 from C:\Users\Peter\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.1.0\cores\arduino/wiring_constants.h:26,
                 from C:\Users\Peter\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.1.0\cores\arduino/wiring.h:34,
                 from C:\Users\Peter\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.1.0\cores\arduino/Arduino.h:36,
                 from sketch\MasterCaution.ino.cpp:1:
c:\users\peter\appdata\local\arduino15\packages\stmicroelectronics\tools\xpack-arm-none-eabi-gcc\10.2.1-1.1\arm-none-eabi\include\c++\10.2.1\bits\stl_algo.h:3474:5: note: candidate: 'template<class _Tp, class _Compare> constexpr _Tp std::min(std::initializer_list<_Tp>, _Compare)'
 3474 |     min(initializer_list<_Tp> __l, _Compare __comp)
      |     ^~~
c:\users\peter\appdata\local\arduino15\packages\stmicroelectronics\tools\xpack-arm-none-eabi-gcc\10.2.1-1.1\arm-none-eabi\include\c++\10.2.1\bits\stl_algo.h:3474:5: note:   template argument deduction/substitution failed:
In file included from C:\Users\Peter\Documents\Arduino\libraries\dcs-bios-arduino-library-0.3.6\src/DcsBios.h:130,
                 from C:\Users\Peter\AppData\Local\Temp\arduino_modified_sketch_800786\MasterCaution.ino:6:
C:\Users\Peter\Documents\Arduino\libraries\dcs-bios-arduino-library-0.3.6\src/internal/RotarySwitch.h:48:54: note:   mismatched types 'std::initializer_list<_Tp>' and 'int'
   48 |     switchValue_ = min(switchValue_+1, maxSwichValue_);
      |                                                      ^
In file included from c:\users\peter\appdata\local\arduino15\packages\stmicroelectronics\tools\xpack-arm-none-eabi-gcc\10.2.1-1.1\arm-none-eabi\include\c++\10.2.1\algorithm:62,
                 from C:\Users\Peter\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.1.0\cores\arduino/wiring_constants.h:26,
                 from C:\Users\Peter\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.1.0\cores\arduino/wiring.h:34,
                 from C:\Users\Peter\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.1.0\cores\arduino/Arduino.h:36,
                 from sketch\MasterCaution.ino.cpp:1:
c:\users\peter\appdata\local\arduino15\packages\stmicroelectronics\tools\xpack-arm-none-eabi-gcc\10.2.1-1.1\arm-none-eabi\include\c++\10.2.1\bits\stl_algo.h:3468:5: note: candidate: 'template<class _Tp> constexpr _Tp std::min(std::initializer_list<_Tp>)'
 3468 |     min(initializer_list<_Tp> __l)
      |     ^~~
c:\users\peter\appdata\local\arduino15\packages\stmicroelectronics\tools\xpack-arm-none-eabi-gcc\10.2.1-1.1\arm-none-eabi\include\c++\10.2.1\bits\stl_algo.h:3468:5: note:   template argument deduction/substitution failed:
In file included from C:\Users\Peter\Documents\Arduino\libraries\dcs-bios-arduino-library-0.3.6\src/DcsBios.h:130,
                 from C:\Users\Peter\AppData\Local\Temp\arduino_modified_sketch_800786\MasterCaution.ino:6:
C:\Users\Peter\Documents\Arduino\libraries\dcs-bios-arduino-library-0.3.6\src/internal/RotarySwitch.h:48:54: note:   mismatched types 'std::initializer_list<_Tp>' and 'int'
   48 |     switchValue_ = min(switchValue_+1, maxSwichValue_);
      |                                                      ^
In file included from c:\users\peter\appdata\local\arduino15\packages\stmicroelectronics\tools\xpack-arm-none-eabi-gcc\10.2.1-1.1\arm-none-eabi\include\c++\10.2.1\algorithm:61,
                 from C:\Users\Peter\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.1.0\cores\arduino/wiring_constants.h:26,
                 from C:\Users\Peter\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.1.0\cores\arduino/wiring.h:34,
                 from C:\Users\Peter\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.1.0\cores\arduino/Arduino.h:36,
                 from sketch\MasterCaution.ino.cpp:1:
c:\users\peter\appdata\local\arduino15\packages\stmicroelectronics\tools\xpack-arm-none-eabi-gcc\10.2.1-1.1\arm-none-eabi\include\c++\10.2.1\bits\stl_algobase.h:278:5: note: candidate: 'template<class _Tp, class _Compare> constexpr const _Tp& std::min(const _Tp&, const _Tp&, _Compare)'
  278 |     min(const _Tp& __a, const _Tp& __b, _Compare __comp)
      |     ^~~
c:\users\peter\appdata\local\arduino15\packages\stmicroelectronics\tools\xpack-arm-none-eabi-gcc\10.2.1-1.1\arm-none-eabi\include\c++\10.2.1\bits\stl_algobase.h:278:5: note:   template argument deduction/substitution failed:
In file included from C:\Users\Peter\Documents\Arduino\libraries\dcs-bios-arduino-library-0.3.6\src/DcsBios.h:130,
                 from C:\Users\Peter\AppData\Local\Temp\arduino_modified_sketch_800786\MasterCaution.ino:6:
C:\Users\Peter\Documents\Arduino\libraries\dcs-bios-arduino-library-0.3.6\src/internal/RotarySwitch.h:48:54: note:   deduced conflicting types for parameter 'const _Tp' ('int' and 'signed char')
   48 |     switchValue_ = min(switchValue_+1, maxSwichValue_);
      |                                                      ^
In file included from c:\users\peter\appdata\local\arduino15\packages\stmicroelectronics\tools\xpack-arm-none-eabi-gcc\10.2.1-1.1\arm-none-eabi\include\c++\10.2.1\algorithm:61,
                 from C:\Users\Peter\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.1.0\cores\arduino/wiring_constants.h:26,
                 from C:\Users\Peter\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.1.0\cores\arduino/wiring.h:34,
                 from C:\Users\Peter\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.1.0\cores\arduino/Arduino.h:36,
                 from sketch\MasterCaution.ino.cpp:1:
c:\users\peter\appdata\local\arduino15\packages\stmicroelectronics\tools\xpack-arm-none-eabi-gcc\10.2.1-1.1\arm-none-eabi\include\c++\10.2.1\bits\stl_algobase.h:230:5: note: candidate: 'template<class _Tp> constexpr const _Tp& std::min(const _Tp&, const _Tp&)'
  230 |     min(const _Tp& __a, const _Tp& __b)
      |     ^~~
c:\users\peter\appdata\local\arduino15\packages\stmicroelectronics\tools\xpack-arm-none-eabi-gcc\10.2.1-1.1\arm-none-eabi\include\c++\10.2.1\bits\stl_algobase.h:230:5: note:   template argument deduction/substitution failed:
In file included from C:\Users\Peter\Documents\Arduino\libraries\dcs-bios-arduino-library-0.3.6\src/DcsBios.h:130,
                 from C:\Users\Peter\AppData\Local\Temp\arduino_modified_sketch_800786\MasterCaution.ino:6:
C:\Users\Peter\Documents\Arduino\libraries\dcs-bios-arduino-library-0.3.6\src/internal/RotarySwitch.h:48:54: note:   deduced conflicting types for parameter 'const _Tp' ('int' and 'signed char')
   48 |     switchValue_ = min(switchValue_+1, maxSwichValue_);
      |                                                      ^
Multiple libraries were found for "Servo.h"
 Used: C:\Users\Peter\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.1.0\libraries\Servo
 Not used: C:\Program Files (x86)\Arduino\libraries\Servo
exit status 1
Error compiling for board Generic STM32F1 series.

talbotmcinnis, 2021-11-10:
I think this will be fixed on master as of a few days ago.  Can you try to pull that down?

speedbird5, 2021-11-10:
Hi, I don't know what you mean.. Sorry.
Are you referring to the Arduino library ?   Can you send me a link ?
Thanks.

talbotmcinnis, 2021-11-10:
https://github.com/talbotmcinnis/dcs-bios-arduino-library/tree/master/src

Download the code from that link, and place it in your documents/arduino/library/dcs-bios... (make sure the files overwrite existing ones to know you are in the right place).

Alternatively I can work on making a new release zip but probably not able until the weekend.

speedbird5, 2021-11-10:
Thanks. I will try that tomorrow when I can and get back to you.

speedbird5, 2021-11-11:
Hi, that worked perfectly for compile and upload.....Thanks.

However, when I use the connect-serial-port.cmd I cannot send switch commands to the aircraft, but it does work with LED light status.

So, it seems that I can't send commands to the sim but the sim can send led light status info back to the board (I'm using the built-in led to test). 

Now, here is the really weird part:  When I use the DCS-BIOS Hub executable to connect to the sim, everything works !!!
Any idea why this may be so ?

Also, is there a way to synchronise the lights in the sim to the physical led lights when we first enter the cockpit?
For example, my built-in LED is on when the Master Caution light in the cockpit if off and vice versa.
I need a way to read the status of all the lights and then set the physical LED lights to that same status.

speedbird5, 2021-11-15:
Hi Talbot, have you had. chance to consider this problem re connection?

talbotmcinnis, 2021-11-18:
Hey.  Very sorry for ghosting you for so long.  Just making sure I understand you:
It sounds like communication from DCS to your board works.  You are saying an LED on your board reacts to DCS?  But a switch on the board does not cause a change in DCS?

I'd start by confirming the serial port can receive.  If you upload a sketch and use the arduino serial monitor (or any serial terminal program), you should be able to toggle the switch and see a human readable line every time the switch state changes.

Next I'd be curious to know if the Chrome Controls Ref works.  You can launch the controls ref, then run DCS and try issuing commands from the control ref and see if they have an effect in DCS.  If not, review your export.lua installation and specifically the files that should be copied along with it.  

In DCS review control bindings for the control you are testing to make sure its not bound to a joystick button that is holding the state in DCS.

You mentioned firewall... are you sending this between PCs or is DCS running on the same PC that the microcontroller is connected to?

Let me know and I'll try to get back to you faster next time!!

speedbird5, 2021-11-18:
Thanks for replying Talbot.

>It sounds like communication from DCS to your board works. You are saying an LED on your board reacts to DCS? But a switch on the board does not cause a change in DCS?
> 

That is correct.

>If you upload a sketch and use the arduino serial monitor (or any serial terminal program), you should be able to toggle the switch and see a human readable line every time the switch state changes.
>

Does the sketch have to have serial.print() statements in it or can it be any sketch and just have  the serial monitor open and flick a switch connected to the board ?

>In DCS review control bindings for the control you are testing to make sure its not bound to a joystick button that is holding the state in DCS.
>

I have no controllers attached during testing with DCS BIOS (except fro rudders).

>You mentioned firewall... are you sending this between PCs or is DCS running on the same PC that the microcontroller is connected to?
>

DCS is running on the same PC that the microcontroller is connected to.  This is not a networked situation.

Why would it work flawlessly with the HUB executable and not the connect-serial-port.cmd ?

I am away from my computer at the moment flying, so I hope to get back to you ASAP with the rest of the testing you asked for.

Regards,  Pete

talbotmcinnis, 2021-11-19:
Great report. It sounds like you have done everything right. My specialty pretty much ends at the microcontrollers so you might need to raise the question back in discord with the dcs-bios side of things.  I have no theories about why hub would work I haven't used it in years.  The original dev ?Ian?  Has been pretty unhelpful about releasing or supporting it afaik but I have never asked him personally.

One idea might be to use portman or performance monitor to check for any conflicting processes using the dcs-bios ports.  You'll see the required ports if you open the script file pointed to by your export lua.  One for out, one for in.  Or maybe increment both sides (socat command and the lua) by the same amount and retry?

speedbird5, 2021-11-19:
Thanks. I’ll try and work out the Portman thing but I think it’s beyond my pay grade.

talbotmcinnis, 2021-11-19:
Portman is beyond what you need actually.  Window resource monitor will show you what you need: http://support.moonpoint.com/os/windows/network/monitoring/resmon/NIS/

Just look for anything on port 7778.

speedbird5, 2021-11-19:
Could Windows 11 be the issue ?

talbotmcinnis, 2021-11-19:
Interesting.  Definitely possible.  Have you tried running socat elevated?

Once socat is running you should definitely see it in listening on 7778.  If not, then something has blocked it.

talbotmcinnis, 2021-11-19:
Sorry, socot sends to 7778 so it’ll be some dcs-bios or dcs itself listening on port 7778.

speedbird5, 2021-11-19:
Yes I have tried running socat  elevated and I get an error. See my earlier post and the error I get.

speedbird5, 2021-11-21:
Talbot, I you have access to a Windows 11 machine, is it possible for you to test ?  thanks

maciekish, 2023-03-02:
Hi, sorry for resurrecting this, is RS485 supported? Its complaining about <libmaple/usart.h>

talbotmcinnis, 2025-04-28:
The core issue of STM32 compilation is closed, so closing this one, but feel free to open new issues for anything outstanding.

closed 2025-04-28 by talbotmcinnis