DCS-Skunkworks/dcs-bios-arduino-library #27: sendDcsBiosMessage sends only boolean values.

Jacktherookie, 2022-07-31:
Hi. I'm trying to send a filtered value (from potentiometer) to DCS BIOS. But when it comes to sendDcsBiosMessage, it only returns boolean value.(sendDcsBiosMessage is defined to return only bool in Dcsbios.h).
Is this correct? If I'm incorrect, is there any way to send integer value to DCS BIOS?

Jacktherookie, 2022-07-31:
When I use default DcsBios::Potentiometer icpHudBrtKnb("ICP_HUD_BRT_KNB", PIN); , this only gets an input directly from PIN. But since Pot has some jittering issues by default, so I'm trying to filter input befor sending to DCS BIOS.
I had to do some extensive googling, and I found this but not working with DCS BIOS Forked version.
https://www.youtube.com/watch?v=dvbB1JNyI54
(DcsBios::PotentiometerEWMA Did not work although PotentiometerEWMA exists in Potentiometer.h.

DeadMeatDCS, 2023-01-10:
Is this still an issue for you?

Filtering is already included in the standard potentiometer class, just using standard filtering parameters set by Ian. See here for explanation of how it works: https://forum.dcs.world/topic/162698-dcs-bios-potentiometer-stutter/?do=findComment&comment=3244436

**PotentiometerEWMA does work**. See the post for what you can do to change the parameters and the result. So if you want to reduce jittering more, you could increase hysteresis to e.g. 256 or more by changing a class like so:

`DcsBios::PotentiometerEWMA<5, 256, 5> icpHudBrtKnb("ICP_HUD_BRT_KNB", PIN);`

You'll lose some precision but for a volume or brightness knob it doesn't matter.

You could also build a lowpass filter to solve it in hardware. Add a capacitor between ground and analogue input.
https://forum.arduino.cc/t/potentiometer-jitter/437767

talbotmcinnis, 2023-08-26:
Seems this issue is dead, and I agree that using existing library solutions is superior to manual intervention.

closed 2023-08-26 by talbotmcinnis