DCS-Skunkworks/dcs-bios-arduino-library #28: Can we get DCS BIOS RS485 going on STM Blue Pills - both master and slave?

virtual812, 2022-08-01:
Hi all,

Been playing with a few devices on the the bench today and got RS485 running, very happy.

At this point the RS485 master requires the device to be an Arduino MEGA2560
Not hard to get them, but they're a big and expensive board.
The STM Blue Pill has 3 or 4 USARTS in addition to the USB port it seems like a great candidate for an RS485 master, more compact and less than 1/4 of the price.

Also would be great to get these going as a slave as well, but not sure how to tell DCS BIOS which USART is connected to a 485 transceiver.

Just a random lookup on ali express, an STM Blue Pill USD $3.33 delivered.
5 / 10 packs are available too.

I'm not great with code, but would be happy to participate  / test code.

OzGadget, 2022-08-01:
pretty please! ;)

virtual812, 2022-09-03:
Anyone had a chance to look at this?

From what I can see there is some code for the 32BIT STM in the  libraries but not being any good with code I can't understand it enough to figure out what to do.

Being able to run multiple busses' off a small, compact, powerful and cheap uC sounds like a great idea.

I'd even go as far as to issue a bounty to someone who could implement and document the use of a BluePill as a standalone USB DCS BIOS unit, a Master for RS485 and as a slave for RS485.

Would this interest anyone?

maciekish, 2023-04-01:
> Anyone had a chance to look at this?
>
> From what I can see there is some code for the 32BIT STM in the libraries but not being any good with code I can't understand it enough to figure out what to [...]

I just posted a bounty for STM32 slaves, maybe you want to join forces so to speak? STM32 master support would be great as well. https://github.com/DCSFlightpanels/dcs-bios-arduino-library/issues/45