DCS-Skunkworks/dcs-bios-arduino-library #38: ATmega32U4 supported as a RS485 Master

kbastronomics, 2022-11-30:
Is it possible to get the ATmega32U4 supported as a RS485 Master?
working on a project we we're using the Sparkfun Pro Micro with HID but would the rest of our pit we'd like to also use it as RS485 Master.

It looks like the issue is with the UART

kbastronomics, 2022-12-27:
Note that I have tried this with VSCode+PlatformIO and Arduino 1.8.6 both give same errors

AerialElectron, 2022-12-28:
If you're comfortable with it, please try giving the "[32U4_Support](https://github.com/AerialElectron/dcs-bios-arduino-library/tree/32U4_Support)" branch of my fork a try. If that works I will submit the pull request for this repo. You will also need to add a `#define UART1_SELECT` to your script and remove the `#define DCS_BIOS_DEFAULT_SERIAL` when you are using RS485. I don't have the hardware to test this out on a live RS485 network, but it does appear to compile correctly now. 
For reference: the root of this issue appears to be that the 32U4 indexes its UART port off of 1 instead of zero. UART0 does not exist, UART1 does. I don't have a deep enough understanding of the RS485 code to comment on any further effects that this might have outside of RS485, but my current implementation essentially just tells BIOS to pretend that it is an ATmega2560 instead of a 32U4 for the purposes of communication. @talbotmcinnis may have a clearer knowledge of what that portion of the code is actually doing.

kbastronomics, 2022-12-28:
Happy to try.
It did compile the slave code see attcahed for one message in it
```
C:\Users\smgvb\Documents\Arduino\libraries\dcs-bios-arduino-library-32U4_Support\src/internal/DcsBiosNgRS485Slave.cpp.inc: In member function 'void DcsBios::RS485Slave::rxISR()':
C:\Users\smgvb\Documents\Arduino\libraries\dcs-bios-arduino-library-32U4_Support\src/internal/DcsBiosNgRS485Slave.cpp.inc:79:12: warning: this statement may fall through [-Wimplicit-fallthrough=]
      state = RX_WAIT_ADDRESS;
      ~~~~~~^~~~~~~~~~~~~~~~~
C:\Users\smgvb\Documents\Arduino\libraries\dcs-bios-arduino-library-32U4_Support\src/internal/DcsBiosNgRS485Slave.cpp.inc:83:4: note: here
    case RX_WAIT_ADDRESS:
    ^~~~

```
[log.txt](https://github.com/DCSFlightpanels/dcs-bios-arduino-library/files/10311788/log.txt)

It does not compile the RS485 Master code though for me (not sure if you address master or not)
here's the log from that compile.

[RS485Master_log.txt](https://github.com/DCSFlightpanels/dcs-bios-arduino-library/files/10311829/RS485Master_log.txt)

I really appreciate your help on this

kbastronomics, 2022-12-28:
I looked at the code around that point and it seems to be expected to fall thru, refering to this
```
C:\Users\smgvb\Documents\Arduino\libraries\dcs-bios-arduino-library-32U4_Support\src/internal/DcsBiosNgRS485Slave.cpp.inc:79:12: warning: this statement may fall through [-Wimplicit-fallthrough=]
      state = RX_WAIT_ADDRESS;
```

AerialElectron, 2022-12-28:
Correct, that fall through is intentional. That particular message is only a warning, so it shouldn't be an issue, but I've corrected it now. 
As for the master version: I'll look into it. Theoretically, it should be possible, but I'm learning this codebase on the fly.

kbastronomics, 2023-02-03:
Been awhile on this issue so checking in to see if there's been any progress on this?

kbastronomics, 2023-08-27:
Kind of gave up on this ask,  I've gotten it compiling now on the Pro-Micro in both Master/Slave modes,   really does come down to the 32U4 has no UART0 that's reserved for the CDC USB port,   so UART1 is the first UART available.   my current code is still untested and have to check the UART BAUD settings.    I'll eventually do a PR once fully tested for it to incorporate if desired

ltshifty, 2024-02-04:
@kbastronomics Did you ever get this working?

I made the mistake of buying ATMEGA32U4 Pro Micro's instead of Nano's, and rather than let them go to waste I'm hoping you've solved this issue.

kbastronomics, 2024-02-04:
Partially,   Slave seems to be working.   master is a little more work.

leomar75, 2024-11-08:
With this current inplementation of RS485 protocol, ATMEGA32U4 cannot be used as Master because of the missing UART0 port.  Rewriting the library to add support Master/Slave is possible, however this is not the route we'd like to take. 
With my commit, I have added minimals changes for enabling RS485 Slave support to ProMicro and Mega boards i.e. AVR_ATmega32U4, AVR_ATmega2560, AVR_ATmega1280.

eHanseJoerg, 2025-06-11:
Based on @leomar75 comment, I suggest to close this issue now as fixed for the slave and 'won't fix" for the master.