DCS-Skunkworks/dcs-bios-arduino-library #39: AH-64D Fire panel Cover

HNR3, 2022-12-24:
This is more of a help request.
Currently working on making all panels for the AH-64D.
I have unfortunatly stumbled upon a problem with the Fire Extg panel.
Is there a way that with a single press of a button i can open the cover then push ENG 1 ?

currently i need to open the PLT_FIRE_ENG1_CVR with my mouse to then press the button on my panel PLT_FIRE_ENG1_BTN.
i have tried binding both the CVR and BTN to 1 pin to see if it would work but it doesnt.

Any of you have tips or work arounds ?

Same used to be with the trigger guard but this got fixed by dcs itself.

any help would be greatly appreciated !!!

AerialElectron, 2022-12-24:
I believe I'm still running a slightly older version of Bios and the Arduino library, but the following should still work:
`DcsBios::SwitchWithCover2Pos eng1FireSuppression("PLT_FIRE_ENG1_BTN", "PLT_FIRE_ENG1_CVR", PIN_NUM);`

This function was introduced to deal with covered switches and buttons. You can find another example in the "OneOfEverything" example file at [https://github.com/DCSFlightpanels/dcs-bios-arduino-library/tree/master/examples/OneOfEverything](url).

Barring that, you may also be able to assign the button to directly operate just the cover. You'd then trigger the button any time the cover status changes to open. I'm not certain that we have any examples of that in the repo, but a quick search on the ED forums should net you some code. This way may also be better if you are using a momentary button rather than a latching/toggle button on your panel.

DeadMeatDCS, 2023-01-10:
@HNR3 is this still an issue for you and can you close it?

The solution by @AerialElectron does work with current version and is applicable for the other covered switches as well, like the APU switch. Just give each instance a name and add the appropriate button and cover names.
`DcsBios::SwitchWithCover2Pos giveItSomeUniqueName("PLT_APU_BTN", "PLT_APU_BTN_CVR", PIN_NUM);`

closed 2023-03-17 by HNR3