DCS-Skunkworks/dcs-bios-arduino-library #40: Arduino Nano Every requires workaround

ColdFrontWI, 2022-12-24:
I have a couple of PCBs that I designed for my Warthog pit (CMSP, CMSC, and ARC-210) that require an Arduino Nano ***Every*** with the ATMega4809 microcontroller.

The boards don't work using RS485, but they do work when directly connected via USB.  However, I have found that I have to first open and close a serial monitor on each board via the Arduino IDE before it will communicate with DCS.  The ports are recognized by socat, but they won't talk to DCS until after the serial monitor trick.

Any idea what might be causing this?  Any suggested alternative approach to make these boards work without the extra step?

Thanks,
Eric

AerialElectron, 2022-12-24:
If this is still an issue, I curious if you are already using, or have tried using `#define DCSBIOS_DEFAULT_SERIAL` rather than IRQ_SERIAL? That would be the easiest fix for a non ATMega328 chip. That being said, I'm not aware of any official support for megaAVR 0 chips in DCS BIOS, or anyone whose tried for that matter, so unless you enjoy being a beta tester it may be best to redesign the boards with an ATMega328 or 2560 and hope you can find some in stock. I'm looking to expand the list of known supported hardware and already have a PCB designed with support for 6-30VDC input, additional 5V power for servos, integrated RS485 and USB, and an onboard LED driver for panel backlighting, but I've yet to get any surface mount ATMega1284s for a decent price to test with. Do give us an update if you got that chip working, the more options DCS BIOS can provide people, the better.

ColdFrontWI, 2023-03-29:
Sorry I missed your comment.

I am able to use the Nanos just fine, and I was already using `#define DCSBIOS_DEFAULT_SERIAL`.  The only trick is that I have to open and close the serial connections _before_ running DCS-BIOS.  To do that I use Realterm and wrote a batch script to automate the process.  One row for each board replacing \<X\> with the port number:
````
realterm port=<X> baud=250000 sendquit
````

As for switching to Nanos, I'd love to but some poor planning on my part makes it complicated.  I designed PCBs for a few of my panels with the Nano in mind, but didn't understand at the time that I couldn't use pins 0 and 1 because they are for the serial connection.  The Every has a separate USB serial connection so pins 0 and 1 work fine.

Thanks,
Eric