DCS-Skunkworks/dcs-bios-arduino-library #41: Bug: Garbled data on complex sketches - (concurrency?) isssue in DcsBiosNgRS485Slave.cpp.inc

krorvik, 2022-12-26:
I am seeing garbled string data (F-16 DED) , and randomly firing LEDs in complex sketches. 

According to this comment from the DCS forums (credit BoboBear):

https://forum.dcs.world/topic/208455-dcs-bios-over-rs485/page/3/#comment-4374711

There seems to be an unreported issue In function:

void RS485Slave::rxISR()

I tested the fix mentioned in the issue, that is, set the state before processing - and this indeed seems to fix the problem. The RX_WAIT_DATA case is changed to look like this:

`
                        case RX_WAIT_DATA:
				rxtx_len--;
				if (rxtx_len == 0) {
					state = RX_WAIT_CHECKSUM;
				}
				if (rx_datatype == RXDATA_DCSBIOS_EXPORT) {
					parser.processCharISR(c);
				}
			break;
`

Should this be considered as a patch?

krorvik, 2022-12-26:
Making this a question for now since I do not have the knowledge to evaluate any other effects.

maciekish, 2023-04-01:
Im seeing the same issue with a slow drawing LCD. Serial buffer too small? Incorrect interrupt handling? Any ideas?