DCS-Skunkworks/dcs-bios-arduino-library #47: TACAN mode selector switch - possible problems with 'new' DCS Dios script

lesthegringo, 2023-06-30:
The behaviour of the rotary switch in game does not match the actual input using the script from the newer DSC Bios documentation,

 I loaded the old script in place of the new, and it works perfectly

New DCS Bios script (problematic)

const byte tacanModePins[9] = {PIN_0, PIN_1, PIN_2, PIN_3, PIN_4}
DcsBios::SwitchMultiPos tacanMode("TACAN_MODE", tacanModePins, 9);
 

Old DSC Bios script (working)

const byte tacanModePins[5] = {PIN_0, PIN_1, PIN_2, PIN_3, PIN_4};
DcsBios::SwitchMultiPos tacanMode("TACAN_MODE", tacanModePins, 5);

WarLord211, 2023-06-30:
we need more info.
Plane/ bios version/ which documentation

lesthegringo, 2023-07-05:
Apologies, should have put that in!

The aircraft is A10C, the faulty DCS Bios is the latest one that I downloaded here https://github.com/DCSFlightpanels/dcs-bios  - I can't see a version number

the correct code comes from the old offline version in HTML form which will not allow me to paste here so all I can do is paste a text copy of the entire TACAN page - sorry if that's not right, however the part of the script that needs changing is the one above

Right Channel SelectorA-10C/TACAN_1
Input Interface: fixed_step Message: TACAN_1 <DEC|INC> Description: switch to previous or next state
DcsBios::RotaryEncoder tacan1("TACAN_1", "DEC", "INC", PIN_A, PIN_B);
Input Interface: action Message: TACAN_1 TOGGLE_XY Description: Toggle TACAN Channel X/Y
DcsBios::ActionButton tacan1ToggleXy("TACAN_1", "TOGGLE_XY", PIN);
Output Type: integer Address: 0x1158 Mask: 0xf000 Shift By: 12 Max. Value: 10 Description: selector position
void onTacan1Change(unsigned int newValue) {
    /* your code here */
}
DcsBios::IntegerBuffer tacan1Buffer(0x1158, 0xf000, 12, onTacan1Change);
Left Channel SelectorA-10C/TACAN_10
Input Interface: fixed_step Message: TACAN_10 <DEC|INC> Description: switch to previous or next state
DcsBios::RotaryEncoder tacan10("TACAN_10", "DEC", "INC", PIN_A, PIN_B);
Output Type: integer Address: 0x1158 Mask: 0x0f00 Shift By: 8 Max. Value: 10 Description: selector position
void onTacan10Change(unsigned int newValue) {
    /* your code here */
}
DcsBios::IntegerBuffer tacan10Buffer(0x1158, 0x0f00, 8, onTacan10Change);
TACAN ChannelA-10C/TACAN_CHANNEL
Output Type: string Address: 0x1162 Max. Length: 4 Description: TACAN Channel
void onTacanChannelChange(char* newValue) {
    /* your code here */
}
DcsBios::StringBuffer<4> tacanChannelBuffer(0x1162, onTacanChannelChange);
TACAN Mode DialA-10C/TACAN_MODE
Input Interface: fixed_step Message: TACAN_MODE <DEC|INC> Description: switch to previous or next state
DcsBios::RotaryEncoder tacanMode("TACAN_MODE", "DEC", "INC", PIN_A, PIN_B);
Input Interface: set_state Message: TACAN_MODE <new_value> Value Range: 0 ... 4 Description: set position
const byte tacanModePins[5] = {PIN_0, PIN_1, PIN_2, PIN_3, PIN_4};
DcsBios::SwitchMultiPos tacanMode("TACAN_MODE", tacanModePins, 5);
Output Type: integer Address: 0x1168 Mask: 0x000e Shift By: 1 Max. Value: 4 Description: selector position
void onTacanModeChange(unsigned int newValue) {
    /* your code here */
}
DcsBios::IntegerBuffer tacanModeBuffer(0x1168, 0x000e, 1, onTacanModeChange);
TACAN Test Indicator LightA-10C/TACAN_TEST
Output Type: integer Address: 0x10da Mask: 0x0400 Shift By: 10 Max. Value: 1 Description: 0 if light is off, 1 if light is on
void onTacanTestChange(unsigned int newValue) {
    /* your code here */
}
DcsBios::IntegerBuffer tacanTestBuffer(0x10da, 0x0400, 10, onTacanTestChange);
DcsBios::LED tacanTest(0x10da, 0x0400, PIN);
TACAN Test ButtonA-10C/TACAN_TEST_BTN
Input Interface: fixed_step Message: TACAN_TEST_BTN <DEC|INC> Description: switch to previous or next state
DcsBios::RotaryEncoder tacanTestBtn("TACAN_TEST_BTN", "DEC", "INC", PIN_A, PIN_B);
Input Interface: set_state Message: TACAN_TEST_BTN <new_value> Value Range: 0 ... 1 Description: set position
const byte tacanTestBtnPins[2] = {PIN_0, PIN_1};
DcsBios::SwitchMultiPos tacanTestBtn("TACAN_TEST_BTN", tacanTestBtnPins, 2);
DcsBios::Switch2Pos tacanTestBtn("TACAN_TEST_BTN", PIN);
Input Interface: action Message: TACAN_TEST_BTN TOGGLE Description: Toggle switch state
DcsBios::ActionButton tacanTestBtnToggle("TACAN_TEST_BTN", "TOGGLE", PIN);
Output Type: integer Address: 0x1158 Mask: 0x0080 Shift By: 7 Max. Value: 1 Description: selector position
void onTacanTestBtnChange(unsigned int newValue) {
    /* your code here */
}
DcsBios::IntegerBuffer tacanTestBtnBuffer(0x1158, 0x0080, 7, onTacanTestBtnChange);
DcsBios::LED tacanTestBtn(0x1158, 0x0080, PIN);
TACAN Signal VolumeA-10C/TACAN_VOL
Input Interface: set_state Message: TACAN_VOL <new_value> Value Range: 0 ... 65535 Description: set the position of the dial
DcsBios::Potentiometer tacanVol("TACAN_VOL", PIN);
Input Interface: variable_step Message: TACAN_VOL <new_value>|-<decrease_by>|+<increase_by> Description: turn the dial left or right
DcsBios::RotaryEncoder tacanVol("TACAN_VOL", "-3200", "+3200", PIN_A, PIN_B);
Output Type: integer Address: 0x1166 Mask: 0xffff Shift By: 0 Max. Value: 65535 Description: position of the potentiometer
void onTacanVolChange(unsigned int newValue) {
    /* your code here */
}
DcsBios::IntegerBuffer tacanVolBuffer(0x1166, 0xffff, 0, onTacanVolChange);
DcsBios::ServoOutput tacanVol(0x1166, PIN, 544, 2400);
TACAN Channel X/Y ToggleA-10C/TACAN_XY
Input Interface: fixed_step Message: TACAN_XY <DEC|INC> Description: switch to previous or next state
DcsBios::RotaryEncoder tacanXy("TACAN_XY", "DEC", "INC", PIN_A, PIN_B);
Input Interface: set_state Message: TACAN_XY <new_value> Value Range: 0 ... 1 Description: set position
const byte tacanXyPins[2] = {PIN_0, PIN_1};
DcsBios::SwitchMultiPos tacanXy("TACAN_XY", tacanXyPins, 2);
DcsBios::Switch2Pos tacanXy("TACAN_XY", PIN);
Input Interface: action Message: TACAN_XY TOGGLE Description: Toggle switch state
DcsBios::ActionButton tacanXyToggle("TACAN_XY", "TOGGLE", PIN);
Output Type: integer Address: 0x1168 Mask: 0x0001 Shift By: 0 Max. Value: 1 Description: selector position
void onTacanXyChange(unsigned int newValue) {
    /* your code here */
}
DcsBios::IntegerBuffer tacanXyBuffer(0x1168, 0x0001, 0, onTacanXyChange);
DcsBios::LED tacanXy(0x1168, 0x0001, PIN);

WarLord211, 2023-07-05:
i looked the HTML ref page and bort booth look good to me like you write -> 5 PINs
[image]
[image]

Maybe i dont get the problem or i misunderstood something.
A screenshot makes it maybe more clear.

lesthegringo, 2023-07-06:
When I click on control reference, this is what I get for A10C - TACAN

WarLord211, 2023-07-08:
?

lesthegringo, 2023-07-08:
can you not see the attachments?

WarLord211, 2023-07-12:
no i dont see an atachment sorry. you can upload the pic direct in the text field

lesthegringo, 2023-07-13:
Hopefully you can see them now - this is the screen I get when I use the
Web interface based BIOS and click on 'control reference in the menu on the
left

[image: DCS BIOS error.jpg]
[image: DCS BIOS error2.jpg]

WarLord211, 2023-07-16:
sorry but the immages dont work for me

talbotmcinnis, 2023-08-26:
I don't understand where you got the example of "New DCS Bios script".  In your first post, that delcares an array of 9 items but only lists 5.  It also passes the array of size 9 into DCS bios.  There are only 5 positions of that switch and 5 pins.

charliefoxtwo, 2025-04-29:
Closing due to inactivity

closed 2025-04-29 by charliefoxtwo