DCS-Skunkworks/dcs-bios-arduino-library #55: Syncronization

kstrance, 2024-01-01:
If possible, am  looking for help in moving from DCSBIOS Hub version to the new FP fork.  Specifically related to syncing physical switches to the game.  The original code worked in HUB and have been porting the new arduino FP library as well.  What it appears is that 'pollinputcurrent' is no longer available in the switches.h library.  I do see the examples of the newer synchronization.ino in the new library, but it appears when trying to update to the newer examples, that 'resetallstates' is not a reference in the library either?  Unfortunately, my current skillset ends there and have spent the better part of two weeks trying to figure out how this works between the HUB version and FP version, changes, etc...  The code pasted below is what was referenced on DCS forums and created by Blue73 that I then used for my INT_LIGHTS panel originally on HUB.  It is driven by the battery switch set to ON and polling the state of switches on that panel. With the latest release and updated arduino FP library, how would this port over to the new version?  Thanks for any help, much appreciated.  -Kye

//INTERIOR LIGHTS//

#define DCSBIOS_RS485_SLAVE 33
#define TXENABLE_PIN 2
#include <DcsBios.h>
#define DCSBIOS_IRQ_SERIAL

//Interior Lights Panel//
DcsBios::Switch3Pos cockkpitLightModeSw("COCKKPIT_LIGHT_MODE_SW", 3, 4);
DcsBios::Switch2Pos lightsTestSw("LIGHTS_TEST_SW", 5);
DcsBios::Potentiometer warnCautionDimmer("WARN_CAUTION_DIMMER", 6);

#define FORCEPOLLCYCLES 10
unsigned int g_iInitIntervalCounter = 0;
unsigned int g_iForcePollCycles = 0;

void setup() {  
  DcsBios::setup();
}

void loop() {
 
  DcsBios::loop();
  
  if ( g_iForcePollCycles > 0 )
   { 
       //repeat poll for this many cycles
       if ( ++g_iInitIntervalCounter == 0 ) 
       {  PollAllControls();     //main loop 1->65535->0 then polls
          g_iForcePollCycles--;
       }
      
   }
   
}

void onBatterySwChange(unsigned int newValue) 
{
     //Battery Switch change, start polling controls FORCEPOLLCYCLES times
     g_iForcePollCycles = FORCEPOLLCYCLES;
}
DcsBios::IntegerBuffer batterySwBuffer(0x54b6, 0x0600, 9, onBatterySwChange);

void PollAllControls()
{
   cockkpitLightModeSw.pollInputCurrent();
   lightsTestSw.pollInputCurrent();
}

jdahlblom, 2024-01-02:
Excellent!

Jocman, 2024-05-20:
@kstrance 
As i'm interessing in syncronization, i was almost in the old procedure, but having some problem to figure out this new one.
Could you have a mail contact with me to help me understand?
Thanks in advance.

kyestrance, 2024-05-21:
@jocman I wish I could help on this one, but @maciekish was the one to truly help.  I am not sure if because I am using his dcsbioskit that utilizes RS485 was originally the issue.  Once the arduino library was updated with the required changes for RS485 and the code was amended to the sketch and uploaded to his boards, everything worked.   I found the syncronization code in the examples folder of the arduino library under EXAMPLES>SYNCRONIZATION>syncronization.ino.  Sorry I cant be more help.

maciekish, 2024-05-21:
@kstrance iirc there was simply a missing function in RS485 mode but it was a while back. Ill see if i can track it down.

Jocman, 2024-05-22:
@kyestrance 
First of all thanks for your answer. I use an (self made) RS485 kit too (about 40 boards - nano and couple of mega - driving switches, encoders, pots, displays, LEDs - tons of LEDs...).
A couple of boring days at work can help a lot, so I studied some libraries and examples and figure out how to make it work.
And it seems to work for me too (at least the experiments I made yesterday evening).
So if there was some issue as you reported at the beginning, I think they fixed it.
No doubt, DCSBIOS team it's a great team.

closed 2025-04-29 by charliefoxtwo