DCS-Skunkworks/dcs-bios-arduino-library #61: Problem uploading / verify code to arduino

BarB310562, 2024-05-25:
Hi
I recently installed DCS Bios and BORT.
I installed the DCS BIOS lib to the Arduino IDE and i made a sketch code.
For unknown reasn to me i can not verify or upload the sketch code to the arduino.
[image]

LeandreArnaud, 2024-06-30:
Hello @BarB310562 
It seems your DCS BIOS arduino lib installation didn't succeed. Did you respects theses steps:
- download .zip of the lib: [here](https://github.com/DCS-Skunkworks/dcs-bios-arduino-library/releases) (current last version: dcs-bios-arduino-library-0.3.9.zip)
- add it to arduino IDE: Sketch / include library / add .zip library / browse for the .zip file

talbotmcinnis, 2025-04-28:
Let use know if LeandreArnaud's instructions cleared this up for you.

closed 2025-04-28 by talbotmcinnis