DCS-Skunkworks/dcs-bios-arduino-library #62: Addresses.h is outdated

charliefoxtwo, 2024-05-29:
The included Addresses.h file is outdated compared to the version in the dcs-bios repo from which it's generated. Additionally, the naming convention for the constants has changed since the current version was last updated, meaning the current Addresses.h effectively will not work with any current bios version.

It also doesn't seem _great_ that we're duplicating this file across repos, but I'm not sure what the best course of action is here.

talbotmcinnis, 2025-04-28:
I agree the duplication is not great.  It has been updated so its... better now, but still need a concept to keep them up to date.

charliefoxtwo, 2025-04-28:
Spitballing...

One option would be to set up a release action, and as part of that action we clone the bios repository and run the lua "compile" code (`lua ./Scripts/DCS-BIOS/test/compile/LocalCompile.lua`) to generate the Addresses.h file, and then attach it to the release. Then it doesn't have to be in this repo anymore. We could use the github release tags so that we're just pulling the latest actual release of dcs-bios.

Of course, to do that we'd have to set up and use release actions in this library, which I don't think we're doing currently. This is also just one option off the top of my head - not saying it's the best option by any means.