DCS-Skunkworks/dcs-bios-arduino-library #67: Question: How to check DCS BIOS Connection state (up/down)

LeandreArnaud, 2024-06-30:
Hello guys ! Thanks you for your work here 👍
I would like to know if there is a way to check if DCS BIOS connection is up or down ? (up = currently live with the plane's data)
I need it in one of my script to do something like this:
```C
if (DcsBios::connectionIsUp() {
  // do whatever
} else {
  // whatever else
}
```

Sorry if it is already in the docs, I didn't find it.

LeandreArnaud, 2024-07-05:
if you are using `DCSBIOS_DEFAULT_SERIAL` (witch is my case) 
I found a way to do it :
```c++
// when dcs bios is up
if (Serial.available()) {
  // do whatever
}
```

Does somebody have a solution for those who use `DCSBIOS_IRQ_SERIAL` ?

kbastronomics, 2024-08-02:
I just use something like this

bool isDCSinmission = false;
char airCravftname[24];

void onAcftNameChange((char* newValue)) {
 if(strsmp(newValue,"")
   isDCSinmission = true;
else
  isDCSinmission = false;
strcpy(newValue,airCraftname);
}
DcsBios::StringBuffer<24> AcftNameBuffer(0x0000, onAcftNameChange);

LeandreArnaud, 2024-08-03:
I think this is a good solution.
Thanks @kbastronomics  :)

closed 2024-08-03 by LeandreArnaud

LeandreArnaud, 2024-08-17:
for those who want a corrected and simplified version:

```c++
bool isDCSinmission = false;

[...]

void onAircraftNameChange(char* newValue) {
  if(strcmp(newValue,""))
    isDCSinmission = true;
  else
    isDCSinmission = false;
}
DcsBios::StringBuffer<24> AcftNameBuffer(0x0000, onAircraftNameChange);
```