DCS-Skunkworks/dcs-bios-arduino-library #68: RS485 and MAX7219

Jocman, 2024-07-13:
### Version

- [X] I am using the latest version

### Aircraft

KA-50 BS3

### Control

----

### Description

I'm using 3 MAX daisychained (2 custom boards: double MAX and a single MAX, daisychained in this order), driven by a nano (only these displays on the nano, nothing else).
If the nano is the only one on the RS485 bus, the displays work correctly (now only some minor issue with the displayed number, only annoyng). But if I connect even just a second board on the bus, the 3rd MAX (the single one) stops working (no display at all - all off).

Initially I tought about a coding issue, to heavy (for the nano and for the way I made it...), but even when I made it lighter (from 700 code lines to about currently 350), i fixed only the freezing problem (even if it was the only one board on RS485 / IRQ SERIAL, in my first attempts after some continuous values changes it freezes; it was really to heavy - to not say stupid -, the microprocessor was getting crazy, I suppose...)

If I switch the displays nano on the IRQ SERIAL and the other boards (by now, in these tests, 4 nano boards, only switches, LEDs, pots and encoders) on RS485, everything works fine: all the displays work smoothly, quickly and correctly (and there's no more the annoying number issue.....).

Considering the tests I made, I can only suppose that the issue is someway related to te RS485 side of DCSBIOS. (sorry, I'm not able to check it)

I've other MAX7219 in my cockpit; without considering the 3 of this issue, there are 3 more but all single, not daisychained, and all wired to their own nano driving also switches, LEDs, and so, and all working. Well, when I tested the panels they worked. There's a 4th one MAX more, but still not tested.

For the sake of honesty, all the tests I'm performing are not really intensive, I mean, I'm testing the panels; they are grouped in 6 stacks, and until now I tested only 4 stacks (individually) but without a "massive" use of DCS, just 10-15 minutes a time, to check if the single stacks works properly. When all stacks will be tested and working, the next step will be to connect all the stacks togheter to RS485 (but the nano of this issue, by now it will run on IRQ SERIAL), and, if everything will work, a massive test.

BTW, here's the link on my posts about the issue (with attached arduino sketches) on DCS forum:
[https://forum.dcs.world/topic/352777-ka50-bs3-dcsbios-and-max7219-displays/](url)

charliefoxtwo, 2024-07-13:
Transferring this to the arduino library repo as this doesn't seem related to core dcs-bios functionality.

talbotmcinnis, 2025-04-28:
I'm wondering if this fix might help.  Can you try out 0.3.11 and add #define DCSBIOS_DEFER_RS485_PROCESSING to your sketch?